AlphaLima/ESP32-Serial-Bridge #3: Communication lost after couple of hours

ginodecock, 2018-05-09:
Hi,

I am using your sketch to remote debug an ESP-32 project.

After a couple of hours it seems that telnet is loosing the connection.

Thanks,
Gino

AlphaLima, 2018-05-09:
Worked for days for me and many other users!
Sonds more like a power/battery problem to me.

ginodecock, 2018-05-10:
Ok, thanks for the reply.I'll see by adding some more capacitors if it will be more stable. Gino

AlphaLima, 2018-05-17:
???
The ESP32 acts as a access point, it cannot initiate a reconnect, must be done by the client.
Have you tried with another ESP32?

ginodecock, 2018-05-17:
Hi,It's pretty firmly solderd to the pcb. Cannot replace it. I use the esp as a wifi station connected to wifi access point. Have you checked this way of working over several hours?Thanks, Gino

???

atsteich, 2018-09-11:
@ginodecock 
are you using bluetooth? If bluetooth is not defined, the index of buf2 will not be resetted. That could explain why it stops after some time, simply because the buffer is full, and not resetted!

AlphaLima, 2018-12-28:
Problem solved after bugfix,

closed 2018-12-28 by AlphaLima