AlphaLima/ESP32-Serial-Bridge #10: Is there a way to make it work @1MBps ?

alternadom, 2020-01-07:
Hi,
I spend a lot of time to make it work at more than the baudrate of 230400 bps.
I need to use it at > 1 000 000 bps but whatever I do, it doesn't :(
Any idea ? Did you make it ?
Thanks for your help!

AlphaLima, 2020-01-07:
No, nerver needed more than 115kBd.
I'm afraid it cant be faster in polling mode. You can try using serial Rx/Tx interrupts together with FIFOs for a higher baudrate.

alternadom, 2020-01-08:
Thanks for your reactivity!
I'm using the Arduino framework but I work on Platformio.. not sure this is changing something :/
Apparently, my problem is not on the serial but on the TCP!
I tried to get (without interrupt) data on Serial(1,3) and put it on Serial(16,17) @ 1Mbps and I'm having NO problems at all..

AlphaLima, 2020-01-08:
OK, 
the data rate depend ton the WiFi Link quality, given that you tried different distances and locations.
The TCP protocol has a lot of data correction overhead, maybe you switch to UDP and increase the packet sizes for a better payload/overhead ratio,

alternadom, 2020-01-08:
Seems that I have just problems from serial to TCP.. not serial to serial and neither TCP to serial :/
The packet size is around 500 and I kept a buffer of 1024..
For my test, i'm using telnet under Macosx, I'm not sure if there is a better solution to test the ESP, if you have something else, I will be glad!
Thanks for your help ;)

alternadom, 2020-01-08:
It appears that TCPClient.write(buf2, i2) is taking too long. But why is that ?!