AlphaLima/ESP32-Serial-Bridge #19: Simultaneously having internet access via mobile data and connect via WLAN to the serial bridge

floriangaller, 2023-01-27:
Hello, 

Im curious if anboby has solved the issue that when you connect via WLAN to the serial brigde internetacces on a normal mobile is not possible anymore. It seems that many mobile deactivate modile date as soon as a WIFI is connected

If anyone has solved that issue it wuold be nice to hear from you

Best regards 

Florian

AlphaLima, 2023-01-27:
Right, thats a logic from android, we cannot change in the serial bridge.
What you can try is:
-Disable mobile data on your Smartphone
-Connect to the ESP (so that android recognizes no internet connection)
-Re-enable mobile data
That how it works for me with a Android Nokia8 Smartphone.
regards
 AL

floriangaller, 2023-02-02:
Thank you for that tip... It works ;-)... But it is strange, because after doing it one time it seams that the phone remembered the setting and now it works without the trick... 

Phone is a Sony H8324

best regards
Florian

closed 2023-02-02 by floriangaller