AlphaLima/ESP32-Serial-Bridge #22: Please add Support for ESP32-S2

fow0ryl, 2024-03-13:
The ESP32-S2 is mostly a replacement for ESP8266, but not supported.
Please add support.

AlphaLima, 2024-05-16:
Sorry I don't have a ESP32-S2 for testing, but it only has two serial ports. Since 3 SIOs is one of the main features of this code, the S2 version has limitations compared to the ESP32. The code should be compatible if you adjust the number an pins of the serial ports.

fow0ryl, 2024-05-16:
Of course, the ESP32-S2 has limitiations. It's more like an ESP8266 than an ESP32.
And changing the number of serial ports is the minor problem. The source code did not compile, because USB is handled completely different. And this requires that the source has to be a mix of ESP8266 and ESP32.
Unable to handle this with only 1 #define ESP32 ...

AlphaLima, 2024-05-16:
The code is designed for an ESP32, not for the ESP32-S2 version!
I see no advantages modifing it for a ESP32-S2 since there is  only two serial ports.