AlphaLima/ESP32-Serial-Bridge #23: Can't change Access Point name

filosy, 2024-05-16:
I tried several times to change the AP name and password in config.h file

const char *ssid = "ABCD";  // You will connect your phone to this Access Point
const char *pw = "glider"; // and this is the password

off course I have save before to load, but when I check if new ssid name is loaded I find always "LK8000"  .
How is it possible?

AlphaLima, 2024-05-16:
Make sure you are actually using the correct file.
Wait for a while until your phone “forgets” the old SSID (at least 5 minutes).
Try using a longer SSID and password (at least 8 characters), following the naming conventions.
Change these one by one to find out if the SSID or password is causing the problem.

filosy, 2024-05-16:
You're right, I used a password of less than 8 characters, after I changed it everything was fine.
Thanks for the help